john30/ebusd-esp32 #93: No status in Ui since new firmware

pascalwinters, 2025-01-28:
Since I updated to the last firmware 2025-01-20 I have issues in the UI. 
Hardware v5 connected through ethernet. 

The adapter works correctly on the background and communicates with the heatpump without any problem.

More info I can't provide, what do you need to start investigating?

See screenshot:

[image]

Koky05, 2025-01-28:
When I open UI I have no option to do anything (whole time is show timeout error with load sign):

[image]

Status from ebusctl:
```
version: ebusd 24.1.24.1
device: 192.168.2.31:9999, TCP, enhanced, firmware 1.1[4c15].1[4c15]
access: *
signal: acquired
```

derjohn, 2025-01-28:
Because @john30 asked in https://github.com/john30/ebusd/issues/1472#issuecomment-2616941008 

...., but ebusd-esp32. thus closing here. please reopen there if needed.
anyway, since I can't reproduce and too little info is given here, please provide more details and check if the settings are still on the device when using the EBUS access point (button click+hold while the leds fade 3 times) ....

I don't get what info the should add? The settings are obvioulsy still there as we can access the thing via Wifi. I didn't change anything other than Wifi Settings. The GUI is still reachable (see screenshot) but can't retrieve /status (probably the API component. Do you want a settings file that we clould copy from somewhere on the device ?

And: All was fine until I did an OTA upgrade to the 202501 version.

derjohn, 2025-01-28:
BTW changelog https://github.com/john30/ebusd-esp32/blob/main/CHANGELOG.md says it "Version 2024-12-21, reported as 1[4c15]". While I cannot see the version in the UI anymore I can remember it was called 202501xx when I clicked on the update. Did some unwanted release sneak in ?

john30, 2025-01-28:
very weird, so it seems that the http server has some issues. didn't recognize anything unusual when i ota updated my testing v5.
could someone connect the adapter to a pc and extract some logging from the serial repl available e.g. via the firmware update page, i.e. enter "log -l 0" in the "logs & console" cmdline or use putty/minicom doing the same?

john30, 2025-01-30:
can someone check which request exactly is done from the browser, e.g. for the /status endpoint?
just tried another v5 with that version but I don't see unusal behaviour with it.
maybe the approach used here helps, please try: https://github.com/john30/ebusd/issues/1477#issuecomment-2622377771

pascalwinters, 2025-01-30:
> can someone check which request exactly is done from the browser, e.g. for the /status endpoint? just tried another v5 with that version but I don't see unusal behaviour with it. maybe the approach used here helps, please try: [john30/ebusd#1477 (comment)](https://github.com/john30/ebusd/issues/1477#issuecomment-2622377771)

This url is requested: http://xxx.xxx.x.xxx/api/v1/status
How can I use the CLI for testing "ota --run"?

pascalwinters, 2025-01-30:
I runned the command "ota --run" through the reply button and now the problem is solved.

[image]

Koky05, 2025-01-30:
In my case it was not success, how can I switch to swap partition via repl?

[image]

SzosszeNET, 2025-01-30:
It takes some time to update normally - think since the GUI gives some feedback what's happening it doesn't feel that long. But yeah CLI runinng ota feels longer, but don't think it is longer.

However - not sure what "progress 0"? Is that the "partition" the updte is on maybe? I'm pretty sure I saw "progress 1" on mine

> At first I was in that it was not working but after some time (cca 10 min) it start to work:
> 
> [image]

SzosszeNET, 2025-01-30:
@john30 if it helps since now understand it's a non-breaking change could try to replicate the issue if it help, just not sure what data I could supply to help understand the problem.

john30, 2025-01-30:
I just tried again, i.e. flashed the 20241221 version, checked that it is running fine, and did an ota update to the recent one. no issues, neither with ebus connectivity now with the http server. really don't get it.
does anyone of you have non-default settings on the "WIFI station/client" section? e.g. a special channel or limited protocols?

anyway, just compiled a new version with more updates from esp-idf and you might want to give it a try by explicitly flashing to it like that:
- point your browser to http://IPADDRESS/?file=testing.bin
- select "UPDATE FIRMWARE"
- select "UPDATE TO SPECIFIC FILE: TESTING.BIN"

SzosszeNET, 2025-01-30:
> I just tried again, i.e. flashed the 20241221 version, checked that it is running fine, and did an ota update to the recent one. no issues, neither with ebus connectivity now with the http server. [...]

Have already resolved by updating from CLI, but flashed the test firmware - completed without any issues.

Not sure what would be non-standard, but don't think anything in here is non-standard?

[image]

derjohn, 2025-01-30:
Requesting http://192.0.2.30/?file=testing.bin redirects me tohttp://192.0.2.30/?file=testing.bin#/status and show the same error. No selector there.

And no, I dont habe anything special in the setup. I just did chose a Wifi on the first install. That's it.

SzosszeNET, 2025-01-30:
> Requesting http://192.0.2.30/?file=testing.bin redirects me tohttp://192.0.2.30/?file=testing.bin#/status and show the same error. No selector there.
> 
> And no, I dont habe anything special in the setup. I just did chose a Wifi on the first install. That's it.

Yes, I think that's expected, the update button would appear on the status section.

I think you can update from cli - by the "Repl" button in the top right corner. To specify the testing.bin I think you want to run "ota --file=testing.bin" or something like that.

derjohn, 2025-01-30:
> It takes some time to update normally - think since the GUI gives some feedback what's happening it doesn't feel that long. But yeah CLI runinng ota feels longer, but don't think it is longer. [...]

That fixed it for me, too. I wasn't aware about the web-cli until I saw the screenshot. Thx.

easi>
Build: unknown
[ebusd device string](https://github.com/john30/ebusd/wiki/2.-Run#device-options): "undefined:/dev/ttyACM0" (number may vary)
WiFi station/client: inactive
WiFi access point: inactive
Ethernet: inactive
Chip ID: unknown,
Hostname: unset
Up time: 0
Free heap: 0 / 0
eBUS: inactive
 REPL
execute: ota --run

OTA: last progress 0, status unknown

shutdown_ebus: shutting down

read_host_tcp: recv: errno 128

close_client: socket closed

close_client: socket closed

shutdown_ebus: proto up

OTA: current: ota_1 ver 20241221

OTA: started

esp-x509-crt-bundle: Certificate validated

OTA: connected

esp_https_ota: Starting OTA...

esp_https_ota: Writing to partition subtype 16 at offset 0x10000

OTA: read desc

OTA: new image: ebusd-esp32 ver 20250120

OTA: verify chip: 5

OTA: read: 1024 / 1402336

OTA: read: 22528 / 1402336

update running: running, progress 2

puni2k, 2025-01-31:
Had this happen to two of my ESPs.

OTA updated my v5 Adapter from 2024-12-21 to 2025-01-20 and it showed the same error. 
ebusd however was able to communicate with it, although the ESP constantly rebooted very frequently.

I then got one of my backup C6 adapters out of storage and OTA updated this as well. Same problem.

Flashing 2025-01-20 via USB worked, luckily.

Field config is:
Wifi Station and AP disabled
Ethernet enabled (USR-ES1)
MQTT enabled

However when I updated the backup C6 Wifi Station was enabled and the Ethernet Shield not connected. Was also not connected to the ebus.

john30, 2025-02-19:
could someone try to reproduce with the new [version 2025-02-16](https://github.com/john30/ebusd-esp32/blob/main/CHANGELOG.md#version-2025-02-16-reported-as-15210)? i.e. flash the old version first and then do the ota to the newest.
I'm still unable to reproduce with v5 and v5-c6 starting from 2024-10-27

john30, 2025-04-29:
ok, thanks, closing then