john30/ebusd-esp32 #104: 1-wire stops working after period of time

devachnid, 2025-09-18:
I have three DS18B20 temperature sensors connected together, and configured. This usually works well, and all three temperatures are visible in the configuration page and also via the API.

However, randomly after a period of time it stops receiving any temperatures (See screenshot below). 

This can only be resolved by a power cycle of the C6 device (a soft reset / reboot doesn't fix it).  There are no relevant errors in the device logs.

Any idea what might be occurring, or how to debug further please?

[image]

```
<html>
<body>
<!--StartFragment-->
22:41:55 |   | [5n
-- | -- | --
22:41:55 | Info | execute: executing commands from '/data/boot.eas'
22:41:55 | Info | execute: strip -m 1 -b 255
22:41:55 | Info | gpio: GPIO[23]\| InputEn: 0\| OutputEn: 1\| OpenDrain: 0\| Pullup: 1\| Pulldown: 0\| Intr:0
22:41:55 | Info | execute: function ledb "@strip -b"
22:41:55 |   | function 'ledb' installed
22:41:55 | Info | execute: function ledg "@strip -g"
22:41:55 |   | function 'ledg' installed
22:41:55 | Info | execute: led -p 16 -i 0 255
22:41:55 |   | 0 duty set to 254
22:41:55 | Info | execute: function ledboot "@led 0"
22:41:55 |   | function 'ledboot' installed
22:41:55 | Info | execute: function leds "@strip -b" "@led 0"
22:41:55 |   | function 'leds' installed
22:41:55 | Info | execute: io -iue 9
22:41:55 |   | IO 9 direction set to 01
22:41:55 |   | IO 9 pull set to 00
22:41:55 |   | IO 9 = 1
22:41:55 | Info | execute: io -iu 6
22:41:55 |   | IO 6 direction set to 01
22:41:55 |   | IO 6 pull set to 00
22:41:55 |   | IO 6 = 1
22:41:55 | Info | execute: nvs_namespace easi
22:41:55 |   | namespace set to 'easi'
22:41:55 | Info | execute: successfully executed 9/9 commands from '/data/boot.eas'
22:41:55 | Info | execute: executing commands from '/data/main.eas'
22:41:55 | Info | execute: var skipstart 0
22:41:55 | Info | execute: leds 255 0 700
22:41:55 | Info | execute: strip -b 255 0 700
22:41:55 | Info | execute: led 0 255 0 700
22:41:55 |   | 0 duty set to 254
22:41:55 | Info | execute: sleep 1000
22:41:55 | Info | main_task: Returned from app_main()
22:41:56 | Info | execute: if "isio -l 9" "var skipstart 1"
22:41:56 | Info | execute: isio -l 9
22:41:56 |   | IO 9 level 1 is unmet
22:41:56 | Info | execute: leds 255 0 700
22:41:56 | Info | execute: strip -b 255 0 700
22:41:56 | Info | execute: led 0 255 0 700
22:41:56 |   | 0 duty set to 254
22:41:56 | Info | execute: sleep 1000
22:41:57 | Info | execute: if "isio -l 9" "var skipstart 1"
22:41:57 | Info | execute: isio -l 9
22:41:57 |   | IO 9 level 1 is unmet
22:41:57 | Info | execute: leds 255 0 700
22:41:57 | Info | execute: strip -b 255 0 700
22:41:57 | Info | execute: led 0 255 0 700
22:41:57 |   | 0 duty set to 254
22:41:57 | Info | execute: sleep 1000
22:41:58 | Info | execute: if "isio -l 9" "var skipstart 1"
22:41:58 | Info | execute: isio -l 9
22:41:58 |   | IO 9 level 1 is unmet
22:41:58 | Info | execute: var rpi 0
22:41:58 | Info | execute: if "var -q stick" else "if \"isio -l 6\" \"var rpi 1\""
22:41:58 | Info | execute: var -q stick
[... 137 lines cut ...]
22:43:04 | Info | OTA: success
22:43:04 | Info | repl_event_task: system event: 12

<!--EndFragment-->
</body>
</html>
```

```
execute: onewire
error: no DS18B20 found
execute: command failed: onewire = 1
```
```
execute: onewire -s
scan completed, found 0 DS18B20
error: no DS18B20 found
execute: command failed: onewire -s = 1
```
```
execute: onewire -s 7
scan completed, found 0 DS18B20
error: no DS18B20 found
execute: command failed: onewire -s 7 = 1
```

john30, 2025-11-01:
please check again with the [new release](https://adapter.ebusd.eu/v5-c6/ChangeLog#20251030) as the 1wire component was updated which e.g. activates a pullup

devachnid, 2025-11-16:
It seems stable at the moment thank you